probe-rs/vscode #10: vscode error Couldn't find debug adapter descriptor

taunusflieger, 2021-09-03:
System: Arch Linux 5.13.13
vscode: 1.60
extension: probe-rs-debugger-0.2.2

When starting debugging the error message "Couldn't find debug adapter descriptor" is shown and no debugging is possible.

launch.jason:

```JSON
{
    "version": "0.2.0",
    "configurations": [
        {
            "type": "probe-rs-debug",
            "request": "launch",
            "name": "probe_rs Executable launch example",
            "cwd": "${workspaceFolder}",
            "program_binary": "target/thumbv7em-none-eabihf/debug/examples/neopixel",
            "chip": "STM32F405RGTx", //!MODIFY
            "connect_under_reset": true,
            "speed": 24000, //!MODIFY
            "probe-index": 0, //!MODIFY
            "runtimeExecutable": "probe-rs-debugger",
            "runtimeArgs": [
                "debug",
                "--dap"
            ],
            "core_index": 0,
            "flashing_enabled": true,
            "reset_after_flashing": true,
            "halt_after_reset": true,
            "console_log_level": "Info", //Error, Warn, Info, Debug 
        }
    ]
}
```
The same error happens with an older version of vscode. I've uninstalled and re-installed the extensions multiple times without any change in behavior.

Any Idea?

noppej, 2021-09-06:
@Yatekii , can you please run this, using the [latest version of the VSCode Extension](https://github.com/probe-rs/vscode/releases/tag/V0.2.2)  against https://github.com/probe-rs/probe-rs/pull/688, with the `console_log_level` at `Debug`, and post the contents of the VSCode *Debug Console*

taunusflieger, 2021-09-06:
In the meantime I received the hs-probe. Which works without this PR well, but with this PR I get the following error:

`
Sent DAP Response: Response {
    body: None,
    command: "attach",
    message: None,
    request_seq: 2,
    seq: 2,
    success: true,
    type_: "response",
}
INFO: FLASHING: Starting write of "/home/michael/src/embedded-rust/feather-f405-neopixel/target/thumbv7em-none-eabihf/debug/feather-f405-neopixel" to device memory
Error while flashing
"Something during the interaction with the core went wrong"	"Something during the interaction with the core went wrong"
"A core architecture specific error occured"		"A core architecture specific error occured"
"Failed to write register DRW at address 0x0000000c"			"Failed to write register DRW at address 0x0000000c"
"An error specific to a probe type occured"				"An error specific to a probe type occured"
"Error handling CMSIS-DAP command Transfer"					"Error handling CMSIS-DAP command Transfer"
"Not enough data in response from probe"						"Not enough data in response from probe"
INFO: Triggered DAP Event: Event {
    body: Some(
        Object({
            "message": String(
                "Error while flashing\n\"Something during the interaction with the core went wrong\"\t\"Something during the interaction with the core went wrong\"\n\"A core architecture specific error occured\"\t\t\"A core architecture specific error occured\"\n\"Failed to write register DRW at address 0x0000000c\"\t\t\t\"Failed to write register DRW at address 0x0000000c\"\n\"An error specific to a probe type occured\"\t\t\t\t\"An error specific to a probe type occured\"\n\"Error handling CMSIS-DAP command Transfer\"\t\t\t\t\t\"Error handling CMSIS-DAP command Transfer\"\n\"Not enough data in response from probe\"\t\t\t\t\t\t\"Not enough data in response from probe\"\n",
            ),
            "severity": String(
                "error",
            ),
        }),
    ),
    event: "probe-rs-show-message",
    seq: 7,
    type_: "event",
}
`

For the STLink V3 I got the following error message:
`
Sent DAP Response: Response {
    body: None,
    command: "attach",
    message: None,
    request_seq: 2,
    seq: 2,
    success: true,
    type_: "response",
}
An error with the usage of the probe occured
"Some functionality was not implemented yet: ST-Links do not support raw SWD access."	"Some functionality was not implemented yet: ST-Links do not support raw SWD access."
INFO: Triggered DAP Event: Event {
    body: Some(
        Object({
            "message": String(
                "An error with the usage of the probe occured\n\"Some functionality was not implemented yet: ST-Links do not support raw SWD access.\"\t\"Some functionality was not implemented yet: ST-Links do not support raw SWD access.\"\n",
            ),
            "severity": String(
                "error",
            ),
        }),
    ),
    event: "probe-rs-show-message",
    seq: 6,
    type_: "event",
}
INFO: Triggered DAP Event: Event {
    body: Some(
        Object({
            "exitCode": Number(
                1,
            ),
        }),
    ),
    event: "exited",
    seq: 8,
    type_: "event",
}
INFO: Closing probe-rs debug extension
`
For Segger edu mini I got:
`
Sent DAP Response: Response {
    body: None,
    command: "attach",
    message: None,
    request_seq: 2,
    seq: 2,
    success: true,
    type_: "response",
}
`
and the probe-rs-debugger reported at the command line:

thread 'main' panicked at 'not yet implemented', probe-rs/src/probe/jlink/swd.rs:1101:9
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace

noppej, 2021-09-09:
This is the result of a problem in `probe-rs`, and is already being tracked as [#793 in that repo](https://github.com/probe-rs/probe-rs/issues/793)