probe-rs/vscode #15: debugging session closes silently

jacobrosenthal, 2021-10-29:
After working through #14 Im trying several target and debugger combinations to try to get a working debug session ust to see what it is supposed to look like. Currently attempting an STM32F407 discovery board with stilnk. But similarly to a jlink the session starts and communicates with the debugger but after a second or two, the session abruptly stops with no feedback. I found the trace debug which is obvious in retrospect (though not listed on the website and probably should be)

* Linux
* vscode 1.61.2
* probe-rs-debugger upstream master 4ba46f40f436e8cec452ce4a5ae570b21d26d156
* vscode upstream master 78242eb09be01398fb3cefe94278d4986fd0543b 

Trace log
https://gist.github.com/jacobrosenthal/a714ba3bc63b7e956f6a94d09701ed80

noppej, 2021-11-15:
@jacobrosenthal Can you please retest this again https://github.com/probe-rs/vscode/releases/tag/v0.3.2 (also please update to the latest `probe-rs-debugger` as suggested in the release note.

PS. I updated the docs to add the `Trace` logging ... thanks for catching that.

jacobrosenthal, 2021-11-15:
Yep still closing silently and nothing with level `Error` interesting printed in the console, 

[image]

though with a daplink instead of the jlink im catching this now, not sure if that was there before, but it also goes away within ~1 second
[image]

noppej, 2021-11-16:
I just received the following update from @jacobrosenthal (only including the relevant parts).
1.  The offending code above uses the embassy-rs project, and apparently runs into https://github.com/probe-rs/probe-rs/issues/531 (with jlink as well as daplink). 
2. When  using an nrf52840 dev kit and nrf-hal example (no embassy) and log level at `Debug`, VSCode pops up the following message
  ```
  ERROR: RangeError [ERR_CHILD_PROCESS_STDIO_MAXBUFFER]: stderr maxBuffer length exceeded [2021-11-16T17:18:53Z INFO probe_rs_debugger::debugger] probe-rs-debugger: ..Starting session from :127.0.0.1:56898 [2021-11-16T17:18:53Z DEBUG probe_rs_debugger::protocol] ...
  ```
3. At a lower log level, like `Error`, it behaves normally (it still suffers from https://github.com/probe-rs/probe-rs/issues/788)

I am proceeding to look at a fix to mitigate # 2 above, so that we can be sure it is not hiding additional info.

noppej, 2021-11-17:
For # 2 above, the [documentation](https://nodejs.org/api/child_process.html#child_processexecfilefile-args-options-callback) states the following:
```text
maxBuffer <number> Largest amount of data in bytes allowed on stdout or stderr. If exceeded, the child process is terminated and any output is truncated. See caveat at maxBuffer and Unicode. Default: 1024 * 1024.
```
The fact that VSCode terminates the childProcess,  explains why the `probe-rs-debugger` logs didn't show the reason for the failure.

commit ea9d587b (2021-11-17): Fix #15: debugging session closes silently
commit 31b1843b (2021-11-17): Merge #23

23: Fix #15: debugging session closes silently r=Tiwalun a=noppej

This fix for #15 includes the following:
- In addition to checking for error `code`, also check for the possibility of a process `signal`. 
- Previously the `probe-rs-debugger` process was started with `child_process.execFile`, which has a default `maxBuffer` for stdio of `1024 * 1024`. When the stdout or stderr (RUST_LOG output is handled this way) exceeded this, VSCode would terminate the process and truncate the output ... and because we didn't test the `signal`, we never saw the `SIGTERM` event happening, and it  [... cut]
closed 2021-11-17 by bors[bot]