probe-rs/vscode #20: Sourcecode cleanup

dbrgn, 2021-11-11:
Would you object to a PR that introduces [prettier](https://prettier.io/) and [eslint](https://eslint.org/) to the codebase? Right now it's a bit of a mess due to no consistent code style (probably because it's based on a plugin template) 🙂

dbrgn, 2021-11-11:
Oops, I forgot to tag the current authors @noppej and @Tiwalun...

Yatekii, 2021-11-11:
Yep, we already discussed this in #16. We would love to have this :)

I am not that familiar with linter in JS/TS, basically all I used in templates from vue/React was kinda terrible. So if you know how to set it up that would be awesome.

Tbh I am not even sure how much we actually have to change in the extension itself, but a formatter is always nice :)

dbrgn, 2021-11-11:
Prettier + eslint + a very strict tsconfig is pretty nice if you want to ensure a clean codebase.

I'll try to come up with a PR, but can't promise any TOA yet 🙂 Feel free to ping me if I should forget about this.

noppej, 2021-11-11:
I think it would be great to enforce some standard here, so please feel free to add it if you have the time :)

Tiwalun, 2021-11-11:
Not at all, I was planning to do the same.

Consistent formatting would be great👍

noppej, 2022-04-12:
My Typescript skills are 'lite', so any PR's for better formatting, or more extensive improvements,  will be appreciated.

dbrgn, 2022-04-12:
Ahh, I totally forgot about this. Maybe I can tackle this issue tonight.

noppej, 2022-04-12:
Awesome :) FYI ... there is a PR #31 that is waiting for review, but I'm guessing we should be able to merge your work into that with reasonable ease.

dbrgn, 2022-04-12:
Hmm, I was busy with other things today, I'll take a note to do it later this week.

dbrgn, 2022-04-13:
Slightly related, is there a specific reason why you're using yarn instead of just going with npm? It's one more dependency that's needed compared to just a plain NodeJS installation.

noppej, 2022-04-13:
Ignorance? LOL. I was new to extensions and followed instructions / examples from the docs ... https://code.visualstudio.com/api/extension-guides/debugger-extension#development-setup-for-mock-debug

If you can achieve the same functionality with one less dependency, then I'm open to being educated :)

closed 2023-01-26 by noppej