probe-rs/vscode #24: An error with the usage of the probe occured

Raunak-Singh-Inventor, 2021-12-23:
I installed the probe-rs vscode extension. Probe-rs detects my chip and my port ,but isn't able to debug my Rust program.

Here is my debug console output:
```
Received DAP Request sequence #1 : initialize
   Sent DAP Response sequence #1 : initialize

Received DAP Request sequence #2 : launch
   Sent DAP Response sequence #2 : launch
ERROR: [2021-12-23T01:46:48Z WARN  probe_rs::config::registry] Found chip nRF52833_xxAA which matches given partial name nRF52833. Consider specifying its full name.

probe-rs-debugger CONSOLE: ....Closing session from  :127.0.0.1:33256
probe-rs-debugger CONSOLE: DAP Protocol server exiting

Triggered DAP Event: probe-rs-show-message
```

Here is my `launch.json`:
```
{
    "version": "0.2.0",
    "configurations": [
        {
            "type": "probe-rs-debug",
            "request": "launch",
            "name": "led_roulette_debugger",
            "cwd": "${workspaceFolder}",
            "programBinary": "/home/raunaksingh/RustProjects/discovery/microbit/target/thumbv7em-none-eabihf/debug/led-roulette.d", //!MODIFY
            "chip": "nRF52833", //!MODIFY
            "connectUnderReset": true,
            "runtimeExecutable": "probe-rs-debugger",
            "runtimeArgs": [
                "debug",
                "--dap"
            ],
            "coreIndex": 0,
            "flashingEnabled": true,
            "resetAfterFlashing": true,
            "haltAfterReset": true,
            "consoleLogLevel": "Info", //Error, Warn, Info, Debug, Trace 
        }
    ]
}
```
I am trying to flash my code to the nRF52833 on the Microbit. I am on Ubuntu Linux.

noppej, 2021-12-23:
Hi.  The clue is in the "ERROR: [2021-12-23T01:46:48Z WARN  probe_rs::config::registry] Found chip nRF52833_xxAA which matches given partial name nRF52833. Consider specifying its full name."

Try changing your launch.json value for `chip` to `nRF52833_xxAA` as suggested   ;-)

Raunak-Singh-Inventor, 2021-12-23:
Changing the chip to `nRF52833_xxAA` removes the warning, but the original error still pops up:
[image]

Here is the updated output to the debug console:
```

Received DAP Request sequence #1 : initialize
   Sent DAP Response sequence #1 : initialize

Received DAP Request sequence #2 : launch
   Sent DAP Response sequence #2 : launch

Triggered DAP Event: probe-rs-show-message
probe-rs-debugger CONSOLE: ....Closing session from  :127.0.0.1:33266
probe-rs-debugger CONSOLE: DAP Protocol server exiting
```

noppej, 2021-12-23:
Sorry about that.  Can you change the `launch.json` value for log to `Debug` and send the results please.

Raunak-Singh-Inventor, 2021-12-23:
Here is the updated output with log value set to Debug:
```
Received DAP Request sequence #1 : initialize
   Sent DAP Response sequence #1 : initialize

Received DAP Request sequence #2 : launch

Sent DAP Response: Response {
    body: None,
    command: "launch",
    message: None,
    request_seq: 2,
    seq: 2,
    success: true,
    type_: "response",
}
ERROR: [2021-12-23T18:34:34Z DEBUG probe_rs::probe::cmsisdap::tools] Found 1 CMSIS-DAP probes using libusb, searching HID
ERROR: [2021-12-23T18:34:34Z DEBUG probe_rs::probe::cmsisdap::tools] Found 1 CMSIS-DAP probes total
ERROR: [2021-12-23T18:34:34Z DEBUG jaylink] libusb 1.0.24.11584
ERROR: [2021-12-23T18:34:34Z DEBUG jaylink] libusb has capability API: true
[2021-12-23T18:34:34Z DEBUG jaylink] libusb has HID access: true
[2021-12-23T18:34:34Z DEBUG jaylink] libusb has hotplug support: true
[2021-12-23T18:34:34Z DEBUG jaylink] libusb can detach kernel driver: true
ERROR: [2021-12-23T18:34:34Z DEBUG probe_rs::probe::cmsisdap::tools] Opening 0d28:0204 in CMSIS-DAPv2 mode
[2021-12-23T18:34:34Z DEBUG probe_rs::probe::cmsisdap::commands] Draining probe of any pending data.
ERROR: [2021-12-23T18:34:34Z DEBUG probe_rs::probe::cmsisdap::commands] Attempt 1 to find packet size
ERROR: [2021-12-23T18:34:34Z DEBUG probe_rs::probe::cmsisdap::commands] Success: packet size is 64
[2021-12-23T18:34:34Z DEBUG probe_rs::probe::cmsisdap::commands] Configuring probe to use packet size 64
ERROR: [2021-12-23T18:34:34Z DEBUG probe_rs::probe::cmsisdap] Detected probe capabilities: Capabilities { _swd_implemented: true, _jtag_implemented: false, swo_uart_implemented: false, swo_manchester_implemented: false, _atomic_commands_implemented: true, _test_domain_timer_implemented: false, swo_streaming_trace_implemented: false, _uart_communication_port_implemented: false, uart_com_port_implemented: false }
ERROR: [2021-12-23T18:34:34Z DEBUG probe_rs::config::registry] Searching registry for chip with name nRF52833_xxAA
ERROR: [2021-12-23T18:34:34Z DEBUG probe_rs::config::registry] Exact match for chip name: nRF52833_xxAA
ERROR: [2021-12-23T18:34:34Z DEBUG probe_rs::probe::cmsisdap] Attaching to target system (clock = 1000kHz)
ERROR: [2021-12-23T18:34:34Z DEBUG probe_rs::probe::cmsisdap] Adding command to batch: Read(port=DebugPort, addr=0)
[2021-12-23T18:34:34Z DEBUG probe_rs::probe::cmsisdap] 1 items in batch
[2021-12-23T18:34:34Z DEBUG probe_rs::probe::cmsisdap] Attempting batch of 1 items
ERROR: [2021-12-23T18:34:34Z DEBUG probe_rs::probe::cmsisdap] 1 of batch of 1 items suceeded
[2021-12-23T18:34:34Z DEBUG probe_rs::architecture::arm::communication_interface] Selecting DP Default
[2021-12-23T18:34:34Z DEBUG probe_rs::architecture::arm::dp] Writing DP register ABORT, value=0x0000001e
[2021-12-23T18:34:34Z DEBUG probe_rs::probe::cmsisdap] Adding command to batch: Write(port=DebugPort, addr=0, data=0x0000001e
[2021-12-23T18:34:34Z DEBUG probe_rs::architecture::arm::dp] Writing DP register SELECT, value=0x00000000
[2021-12-23T18:34:34Z DEBUG probe_rs::probe::cmsisdap] Adding command to batch: Write(port=DebugPort, addr=8, data=0x00000000
[2021-12-23T18:34:34Z DEBUG probe_rs::architecture::arm::dp] Reading DP register CTRL/STAT
[2021-12-23T18:34:34Z DEBUG probe_rs::probe::cmsisdap] Adding command to batch: Read(port=DebugPort, addr=4)
[2021-12-23T18:34:34Z DEBUG probe_rs::probe::cmsisdap] 3 items in batch
[2021-12-23T18:34:34Z DEBUG probe_rs::probe::cmsisdap] Attempting batch of 3 items
ERROR: [2021-12-23T18:34:34Z DEBUG probe_rs::probe::cmsisdap] 3 of batch of 3 items suceeded
[2021-12-23T18:34:34Z DEBUG probe_rs::architecture::arm::dp] Read    DP register CTRL/STAT, value=0x00000000
[2021-12-23T18:34:34Z DEBUG probe_rs::architecture::arm::dp] Writing DP register CTRL/STAT, value=0x50000000
[2021-12-23T18:34:34Z DEBUG probe_rs::probe::cmsisdap] Adding command to batch: Write(port=DebugPort, addr=4, data=0x50000000
[2021-12-23T18:34:34Z DEBUG probe_rs::architecture::arm::dp] Reading DP register CTRL/STAT
[2021-12-23T18:34:34Z DEBUG probe_rs::probe::cmsisdap] Adding command to batch: Read(port=DebugPort, addr=4)
[2021-12-23T18:34:34Z DEBUG probe_rs::probe::cmsisdap] 2 items in batch
[2021-12-23T18:34:34Z DEBUG probe_rs::probe::cmsisdap] Attempting batch of 2 items
ERROR: [2021-12-23T18:34:34Z DEBUG probe_rs::probe::cmsisdap] 2 of batch of 2 items suceeded
[2021-12-23T18:34:34Z DEBUG probe_rs::architecture::arm::dp] Read    DP register CTRL/STAT, value=0xf0000000
[2021-12-23T18:34:34Z DEBUG probe_rs::architecture::arm::dp] Writing DP register CTRL/STAT, value=0x50000f00
[2021-12-23T18:34:34Z DEBUG probe_rs::probe::cmsisdap] Adding command to batch: Write(port=DebugPort, addr=4, data=0x50000f00
[2021-12-23T18:34:34Z DEBUG probe_rs::architecture::arm::dp] Reading DP register CTRL/STAT
[2021-12-23T18:34:34Z DEBUG probe_rs::probe::cmsisdap] Adding command to batch: Read(port=DebugPort, addr=4)
[2021-12-23T18:34:34Z DEBUG probe_rs::probe::cmsisdap] 2 items in batch
[2021-12-23T18:34:34Z DEBUG probe_rs::probe::cmsisdap] Attempting batch of 2 items
ERROR: [2021-12-23T18:34:34Z DEBUG probe_rs::probe::cmsisdap] 2 of batch of 2 items suceeded
[2021-12-23T18:34:34Z DEBUG probe_rs::architecture::arm::dp] Read    DP register CTRL/STAT, value=0xf0000f00
[2021-12-23T18:34:34Z DEBUG probe_rs::architecture::arm::dp] Reading DP register CTRL/STAT
[2021-12-23T18:34:34Z DEBUG probe_rs::probe::cmsisdap] Adding command to batch: Read(port=DebugPort, addr=4)
[... cut ...]
ERROR: [2021-12-23T18:34:34Z DEBUG probe_rs_debugger::protocol] log_to_console
probe-rs-debugger CONSOLE: ....Closing session from  :127.0.0.1:33268
probe-rs-debugger CONSOLE: DAP Protocol server exiting
INFO: Triggered DAP Event: Event {
    body: Some(
        Object({
            "message": String(
                "An error with the usage of the probe occured\n",
            ),
            "severity": String(
                "error",
            ),
        }),
    ),
    event: "probe-rs-show-message",
    seq: 7,
    type_: "event",
}

```

noppej, 2021-12-27:
Hi. I am testing against a MicroBit 2.0 board, using the same `launch.json` settings as you, and am not able to reproduce the error. Are you able to share your source (or at least your `Cargo.toml` and a small `.rs` that reproduces the problem so I can try it out here?

Raunak-Singh-Inventor, 2021-12-27:
Here is the project:
[https://github.com/Raunak-Singh-Inventor/05-led-roulette-discovery](https://github.com/Raunak-Singh-Inventor/05-led-roulette-discovery)

noppej, 2021-12-27:
I can reproduce with your code, and then get it to work with one small change to your `launch.json`. I should have spotted this sooner :)

Please remove the `.d` from the end of the `programBinary` option. The binary file is simply named `led-roulette` without any file extensions. 

If you can confirm this works for you, then I will open a PR to generate a more useful error message.

Raunak-Singh-Inventor, 2021-12-27:
@noppej Removing `.d` from the end of programBinary in the launch.json doesn't fix the issue. I updated the launch.json in the repo.

noppej, 2021-12-27:
Can you try to set `connectUnderReset` to `false` please.

Raunak-Singh-Inventor, 2021-12-27:
@noppej  Setting `connectUnderReset` to `false` creates a new error:
[image]
Here is my output:
https://gist.github.com/Raunak-Singh-Inventor/805d230dc921bb2c03d231a61cdf0737

noppej, 2021-12-27:
OK, that means you are debugging. The message you are getting is a known limitation. If you set a breakpoint at a valid line of RUST code in your program, the debugger will halt there as expected. 

I will fix the original problem reported by providing a suitable error message when trying to flash an invalid binary file.

closed 2022-01-01 by Raunak-Singh-Inventor