probe-rs/vscode #26: defmt::println over RTT not shown in terminal

darryln, 2022-01-16:
`defmt::println!("Hello, world!")` does not produce output in the terminal window.

[Code to repro.](https://github.com/darryln/hello-probe-rs/tree/hello-defmt)  

hardware: 

- nucleo-F767ZI powered over usb, nothing attached to pins.

software: 

- probe-rs-debugger-0.3.3.vsix, probe-rs-debugger 0.12.0, probe-run 0.3.1

noppej, 2022-01-18:
@darryln Hi, can you please add ` "rttEnabled": true` to your `launch.json` and try again. 

The default for this setting is false, even if you have defmt in your code.

darryln, 2022-01-18:
Added "rttEnabled" = true, still no output.   

note that 'cargo run' _does_ show the output.  'cargo tree' output included below.

Running `probe-run --chip STM32F767ZITx target/thumbv7em-none-eabihf/debug/hello`
(HOST) INFO  flashing program (7 pages / 7.00 KiB)
(HOST) INFO  success!
────────────────────────────────────────────────────────────────────────────────
Hello, world!
└─ hello::__cortex_m_rt_main @ src/main.rs:8
────────────────────────────────────────────────────────────────────────────────
(HOST) INFO  device halted without error
embsysdev@silvio:~/rust/embedded/hello-probe-rs$ 

$ cargo tree
hello v0.1.0 (/home/embsysdev/rust/embedded/hello-probe-rs)
├── cortex-m v0.7.4
│   ├── bare-metal v0.2.5
│   │   [build-dependencies]
│   │   └── rustc_version v0.2.3
│   │       └── semver v0.9.0
│   │           └── semver-parser v0.7.0
│   ├── bitfield v0.13.2
│   ├── embedded-hal v0.2.6
│   │   ├── nb v0.1.3
│   │   │   └── nb v1.0.0
│   │   └── void v1.0.2
│   └── volatile-register v0.2.1
│       └── vcell v0.1.3
├── cortex-m-rt v0.7.1
│   └── cortex-m-rt-macros v0.7.0 (proc-macro)
│       ├── proc-macro2 v1.0.36
│       │   └── unicode-xid v0.2.2
│       ├── quote v1.0.14
│       │   └── proc-macro2 v1.0.36 (*)
│       └── syn v1.0.85
│           ├── proc-macro2 v1.0.36 (*)
│           ├── quote v1.0.14 (*)
│           └── unicode-xid v0.2.2
├── defmt v0.3.0
│   ├── bitflags v1.3.2
│   └── defmt-macros v0.3.1 (proc-macro)
│       ├── defmt-parser v0.3.0
│       ├── proc-macro-error v1.0.4
│       │   ├── proc-macro-error-attr v1.0.4 (proc-macro)
│       │   │   ├── proc-macro2 v1.0.36 (*)
│       │   │   └── quote v1.0.14 (*)
│       │   │   [build-dependencies]
│       │   │   └── version_check v0.9.4
│       │   ├── proc-macro2 v1.0.36 (*)
│       │   ├── quote v1.0.14 (*)
│       │   └── syn v1.0.85 (*)
│       │   [build-dependencies]
│       │   └── version_check v0.9.4
│       ├── proc-macro2 v1.0.36 (*)
│       ├── quote v1.0.14 (*)
│       └── syn v1.0.85 (*)
├── defmt-rtt v0.3.1
│   ├── cortex-m v0.7.4 (*)
│   └── defmt v0.3.0 (*)
├── panic-probe v0.3.0
│   ├── cortex-m v0.7.4 (*)
│   └── defmt v0.3.0 (*)
└── stm32f7xx-hal v0.6.0 (/home/embsysdev/rust/embedded/stm32f7xx-hal)
    ├── as-slice v0.2.1
    │   └── stable_deref_trait v1.2.0
    ├── bare-metal v1.0.0
    ├── cast v0.3.0
    ├── cortex-m v0.7.4 (*)
    ├── cortex-m-rt v0.7.1 (*)
    ├── embedded-hal v0.2.6 (*)
    ├── embedded-time v0.12.1
    │   └── num v0.3.1
    │       ├── num-complex v0.3.1
    │       │   └── num-traits v0.2.14
    │       │       [build-dependencies]
    │       │       └── autocfg v1.0.1
    │       ├── num-integer v0.1.44
    │       │   └── num-traits v0.2.14 (*)
    │       │   [build-dependencies]
    │       │   └── autocfg v1.0.1
    │       ├── num-iter v0.1.42
    │       │   ├── num-integer v0.1.44 (*)
    │       │   └── num-traits v0.2.14 (*)
    │       │   [build-dependencies]
    │       │   └── autocfg v1.0.1
    │       ├── num-rational v0.3.2
    │       │   ├── num-integer v0.1.44 (*)
    │       │   └── num-traits v0.2.14 (*)
    │       │   [build-dependencies]
    │       │   └── autocfg v1.0.1
    │       └── num-traits v0.2.14 (*)
    ├── micromath v2.0.0
    ├── nb v1.0.0
    ├── rand_core v0.6.3
    ├── stm32-fmc v0.2.4
    │   └── embedded-hal v0.2.6 (*)
    ├── stm32f7 v0.14.0
    │   ├── bare-metal v1.0.0
    │   ├── cortex-m v0.7.4 (*)
    │   ├── cortex-m-rt v0.7.1 (*)
    │   └── vcell v0.1.3
    ├── time v0.3.5
    └── void v1.0.2
[dev-dependencies]
└── defmt-test v0.3.0
    ├── cortex-m v0.7.4 (*)
    ├── cortex-m-rt v0.7.1 (*)
    ├── defmt v0.3.0 (*)
    └── defmt-test-macros v0.3.0 (proc-macro)
        ├── proc-macro2 v1.0.36 (*)
        ├── quote v1.0.14 (*)
        └── syn v1.0.85 (*)

Yatekii, 2022-01-19:
This is most likely because you are using defmt 0.3 when probe-rs-debugger is using 0.2.

noppej, 2022-01-19:
Actually, I suspect it has to do with the `cortex_m::asm::bkpt();` that gets executed as part of the main loop. 

Can you replace that with at `cortex_m::asm::nop();` and see if that changes things? The debugger won't poll for RTT while the target is halted at a breakpoint.

Yatekii, 2022-01-19:
Yep, that for sure is an issue too :)
But once that is resolved, defmt will for sure spew garbled up data because of the incompatible versions :)

darryln, 2022-01-19:
changing from bkpt to nop() has no effect.

darryln, 2022-01-19:
Is there a hello world example with all the dependencies resolved that has been verified?

noppej, 2022-01-19:
I have a private one.  Will send it in the morning.

darryln, 2022-01-19:
Excellent!  Will this be a point release of the vsix?

noppej, 2022-01-19:
Yes, and of the `probe-rs-debugger`. I will send you links when everything is ready for you to test.

noppej, 2022-01-19:
@darryln Please can you test this with the combination of the attached VSIX file, as well as the special branch of `probe-rs-debugger` at 
```
cargo install --git https://github.com/probe-rs/probe-rs probe-rs-debugger --branch noppej/issue894
```
[probe-rs-debugger-0.3.4.vsix.zip](https://github.com/probe-rs/vscode/files/7900579/probe-rs-debugger-0.3.4.vsix.zip)

noppej, 2022-01-19:
@darryln Also, as promised, you can look at the sample app `stm32h745-defmt-rtt` in [this-repo](https://github.com/noppej/probe-rs-debugger-test) for a recently tested example of using `defmt` on an STM32 board.

darryln, 2022-01-19:
@noppej verified!
[image]

darryln, 2022-01-19:
@Yatekii @noppej also, this works.
```
pub fn exit() -> ! {
    defmt::println!("exit");
    loop {
        cortex_m::asm::bkpt();
        //cortex_m::asm::nop();
    }
}
```

noppej, 2022-01-19:
Nice. I appreciate your patience and willingness to help verify. Thanks :)

darryln, 2022-01-19:
@noppej Well...sad to report but now getting no output, same as before. Restarted vscode, no change.  I noticed that the defmt terminal instance does not get created.  probe-run still works correctly.

noppej, 2022-01-20:
Did anything change since the last time it worked?

Have you tried disconnecting the board and reconnecting again?

darryln, 2022-01-20:
Yes, unplugged board from usb.  Added the default rttChannelFormats to launch.json then reverted - no effect either way. 

SWAG: maybe some VS internal flag is persisted with the session?

noppej, 2022-01-20:
@darryln Two quick comments
- The source file path is provided by the binary file (ELF format) and not much we can do about when it is wrong. In addition to this case, I've also seen it when the breakpoint happens in the RUST compiler builtins.  Other debuggers will struggle with the same problem. The best we can do is to show assembly code for these cases. That is on our wish-list, and happy to accept PR's for that :)
- I will implement some additional error handling on the RTT to see if we can identify the root cause of why your RTT output gets lost. Watch this space for an update ...

darryln, 2022-01-20:
Re source file path: objdump shows multiple references to lib.rs and the debugger (or dependency that parses the obj) is prepending the wrong base path to asm/lib.rs. The following entry shows the base path as "./" but the bogus path mentioned above (which does not exist) is generated, so it's substituting the base path to my source.  The second entry below is the reference to src/lib.rs in _my_ crate, which has the correct base path.  So it looks like a bug (somewhere). Should I file a separate issue?

```
 <1><b8201>: Abbrev Number: 0
  Compilation Unit @ offset 0xb8202:
   Length:        0xa09 (32-bit)
   Version:       4
   Abbrev Offset: 0xd59
   Pointer Size:  4
 <0><b820d>: Abbrev Number: 1 (DW_TAG_compile_unit)
    <b820e>   DW_AT_producer    : (indirect string, offset: 0xb9288): clang LLVM (rustc version 
1.59.0-nightly (c5ecc1570 2021-12-15))
    <b8212>   DW_AT_language    : 28    (Rust)
    <b8214>   DW_AT_name        : (indirect string, offset: 0xe1171): asm/lib.rs
    <b8218>   DW_AT_stmt_list   : 0x5599a
    <b821c>   DW_AT_comp_dir    : (indirect string, offset: 0xb075b): ./
    <b8220>   DW_AT_GNU_pubnames: 1
    <b8220>   DW_AT_low_pc      : 0x0
    <b8224>   DW_AT_ranges      : 0xdf80

```

```
  Compilation Unit @ offset 0xf3:
   Length:        0xf0 (32-bit)
   Version:       4
   Abbrev Offset: 0x87
   Pointer Size:  4
 <0><fe>: Abbrev Number: 1 (DW_TAG_compile_unit)
    <ff>   DW_AT_producer    : (indirect string, offset: 0x6a439): clang LLVM (rustc version 1.5
8.0 (02072b482 2022-01-11))
    <103>   DW_AT_language    : 28      (Rust)
    <105>   DW_AT_name        : (indirect string, offset: 0x9a1b9): src/lib.rs
    <109>   DW_AT_stmt_list   : 0xf6
    <10d>   DW_AT_comp_dir    : (indirect string, offset: 0x39311): /home/embsysdev/rust/embedded/hello-probe-rs
    <111>   DW_AT_GNU_pubnames: 1
    <111>   DW_AT_low_pc      : 0x0
    <115>   DW_AT_ranges      : 0x18
 <1><119>: Abbrev Number: 2 (DW_TAG_namespace)
    <11a>   DW_AT_name        : (indirect string, offset: 0x2969f): hello
 <2><11e>: Abbrev Number: 2 (DW_TAG_namespace)
    <11f>   DW_AT_name        : (indirect string, offset: 0x5a2f6): exit
 <3><123>: Abbrev Number: 3 (DW_TAG_variable)

```

noppej, 2022-01-20:
Yes, please log it under probe-rs/probe-rs repo, and include the above. It should be easily fixable.

commit b751ff23 (2022-02-03): Merge #27

27: Various fixes and updates. Requires updated `probe-rs-debugger` r=Yatekii a=noppej

- Adapt to `defmt` 0.3 'Rzcobs' encoding to fix #26.
- Support the new `defmt` 0.3 `DEFMT_LOG` environment variable.
- Catch target-side `log::error!()` and display as client-side message.

** Depends on the changes in [probe-rs-debugger PR #895](https://github.com/probe-rs/probe-rs/pull/895)**
```
cargo install --git https://github.com/probe-rs/probe-rs probe-rs-debugger --branch noppej/issue894
```
PR #27 (2022-02-03): Various fixes and updates. Requires updated `probe-rs-debugger`
- Adapt to `defmt` 0.3 'Rzcobs' encoding to fix #26.
- Support the new `defmt` 0.3 `DEFMT_LOG` environment variable.
- Catch target-side `log::error!()` and display as client-side message.

** Depends on the changes in [probe-rs-debugger PR #895](https://github.com/probe-rs/probe-rs/pull/895)**
```
cargo install --git https://github.com/probe-rs/probe-rs probe-rs-debugger --branch noppej/issue894
```
closed 2022-02-03 by bors[bot]