probe-rs/vscode #36: Debugging Session Errors on Breakpoint

curiousmuch, 2022-07-03:
Hello! 

The first break point I hit debugging my application results in a number of errors in the debugging log. I'm using the latest version probe-rs w/ the latest release version of the vscode extension. 
 - The program does seem to load correctly and run correctly...
 - I don't quite understand these memory locations the debugger is attempting to access based on my linker script. 
```
MEMORY
{
    FLASH : ORIGIN = 0x8000000, LENGTH = 1024K
    RAM : ORIGIN = 0x24000000, LENGTH = 128K
}
```
*Example 1:*
```
[2022-07-03T18:26:42Z ERROR probe_rs::debug::debug_info] Failed to read referenced variable address from memory location 0x88000024 : An error with the usage of the probe occurred.
```
My flash only runs from 0x08000000 - 0x080FFFFF 

*Example 2:*
```
[2022-07-03T18:26:42Z ERROR probe_rs::debug::debug_info] Failed to read referenced variable address from memory location 0x1CCE580E : An error with the usage of the probe occurred.
```
I'm clueless where this is suppose to be lol. 

**VSCode Launch File**
```
{
    "version": "0.2.0",
    "configurations": [
        {
            "type": "probe-rs-debug",
            "request": "launch",
            "name": "probe_rs Executable launch example",
            "cwd": "${workspaceFolder}",
            "connectUnderReset": true,
            "speed": 24000, //!MODIFY (or remove)
            // "probe": "PID:VID:<Serial>", //!MODIFY (or remove)
            "runtimeExecutable": "probe-rs-debugger",
            "runtimeArgs": [
                "debug"
            ],
            "flashingConfig": {
                "flashingEnabled": true,
                "resetAfterFlashing": true,
                "haltAfterReset": true,
            },
            "chip": "STM32H723ZGTx", //!MODIFY
            "coreConfigs": [
                {
                    "coreIndex": 0,
                    "programBinary": "target/thumbv7em-none-eabihf/debug/blinky", //!MODIFY
                    // "svdFile": "Relative or fully qualified path to your programBinary", //!MODIFY
                }
            ],
            "consoleLogLevel": "Info", //Error, Warn, Info, Debug, Trace 
        }
    ]
}
```

**Debugging Log File. **
```
Received DAP Request sequence #1 : initialize
   Sent DAP Response sequence #1 : initialize

Received DAP Request sequence #2 : launch
INFO: FLASHING: Starting write of "/home/curiousmuch/Repos/Rust/embassy/examples/stm32h7/target/thumbv7em-none-eabihf/debug/blinky" to device memory

Triggered DAP Event: progressStart

[... 295 lines cut ...]
[2022-07-03T18:26:42Z ERROR probe_rs::debug::debug_info] Failed to read referenced variable address from memory location 0x2E5BF0DD : An error with the usage of the probe occurred.
[2022-07-03T18:26:42Z ERROR probe_rs::debug::debug_info] Failed to read referenced variable address from memory location 0x2E5BF0E1 : An error with the usage of the probe occurred.
[2022-07-03T18:26:42Z ERROR probe_rs::debug::debug_info] Failed to read referenced variable address from memory location 0x2E5BF0E5 : An error with the usage of the probe occurred.
[2022-07-03T18:26:42Z WARN  probe_rs::probe::stlink] send_jtag_command 242 failed: SwdApFault
[2022-07-03T18:26:42Z ERROR probe_rs::debug::debug_info] Failed to read referenced variable address from memory location 0x2E5BF0E9 : An error with the usage of the probe occurred.
[2022-07-03T18:26:42Z WARN  probe_rs::probe::stlink] send_jtag_command 242 failed: SwdApFault
[2022-07-03T18:26:42Z ERROR probe_rs::debug::debug_info] Failed to read referenced variable address from memory location 0x1CCE580E : An error with the usage of the probe occurred.
[2022-07-03T18:26:42Z ERROR probe_rs::debug::debug_info] Failed to read referenced variable address from memory location 0x1CCE5812 : An error with the usage of the probe occurred.
[2022-07-03T18:26:42Z ERROR probe_rs::debug::debug_info] Failed to read referenced variable address from memory location 0x1CCE5816 : An error with the usage of the probe occurred.
[2022-07-03T18:26:42Z WARN  probe_rs::probe::stlink] send_jtag_command 242 failed: SwdApFault
[2022-07-03T18:26:42Z ERROR probe_rs::debug::debug_info] Failed to read referenced variable address from memory location 0x1CCE581A : An error with the usage of the probe occurred.
[2022-07-03T18:26:42Z ERROR probe_rs::debug::debug_info] Failed to read referenced variable address from memory location 0x7CD10585 : An error with the usage of the probe occurred.
[2022-07-03T18:26:42Z ERROR probe_rs::debug::debug_info] Failed to read referenced variable address from memory location 0x7CD10589 : An error with the usage of the probe occurred.
[2022-07-03T18:26:42Z ERROR probe_rs::debug::debug_info] Failed to read referenced variable address from memory location 0x7CD1058D : An error with the usage of the probe occurred.
[2022-07-03T18:26:42Z ERROR probe_rs::debug::debug_info] Failed to read referenced variable address from memory location 0x7CD10591 : An error with the usage of the probe occurred.
[2022-07-03T18:26:42Z WARN  probe_rs::probe::stlink] send_jtag_command 242 failed: SwdApWdataError
[2022-07-03T18:26:42Z WARN  probe_rs::probe::stlink] send_jtag_command 242 failed: SwdApFault
[2022-07-03T18:26:42Z ERROR probe_rs::debug::debug_info] Failed to read referenced variable address from memory location 0x2E5BF0DD : An error with the usage of the probe occurred.
[2022-07-03T18:26:42Z ERROR probe_rs::debug::debug_info] Failed to read referenced variable address from memory location 0x2E5BF0E1 : An error with the usage of the probe occurred.
[2022-07-03T18:26:42Z ERROR probe_rs::debug::debug_info] Failed to read referenced variable address from memory location 0x2E5BF0E5 : An error with the usage of the probe occurred.
[2022-07-03T18:26:42Z WARN  probe_rs::probe::stlink] send_jtag_command 242 failed: SwdApFault
[2022-07-03T18:26:42Z ERROR probe_rs::debug::debug_info] Failed to read referenced variable address from memory location 0x2E5BF0E9 : An error with the usage of the probe occurred.

Received DAP Request sequence #17 : variables
   Sent DAP Response sequence #17 : variables

noppej, 2022-07-05:
Thanks. It definitely seems like there is something funky with the memory addresses. These are calculated using the debug info (DWARF) generated as part of creating the binary, and it is quite possible that there is an issue with the logic.

Is this just the standard 'blinky' example? Have you made any changes to the config files? Can you give me some info on where you've set the breakpoint in the source file? 

I don't get these errors, so to reproduce/fix I will need a little more info please.

noppej, 2022-08-23:
@curiousmuch I've done some work against breakpoints and variable handling that is now checked into the 'master' branch. Any chance you can retest, and/or answer some of the above questions?

noppej, 2022-09-12:
Closing until further information/testing by OP.

closed 2022-09-12 by noppej