probe-rs/vscode #40: "probe-rs debugger process encountered an error" when trying to launch

tgross35, 2023-01-25:
Hello everyone,

I haven't had any success launching the VScode launch file. I get the following error:

```
probe-rs-debugger process encountered an error: {"errno":-2,"code":"ENOENT","syscall":"spawn probe-rs-debugger","path":"probe-rs-debugger","spawnargs":["debug","--port","54166","--vscode"]}
```

The launch config:

```json5
        {
            "name": "probe-rs Debug TestDebug",
            // "type": "probe-rs-debug",
            "preLaunchTask": "rust: cargo build test-sam",
            "request": "launch",
            "cwd": "${workspaceFolder}",
            "connectUnderReset": true,
            // "chip": "STM32H745ZITx",
            "flashingConfig": {
                "flashingEnabled": true,
                "resetAfterFlashing": true,
                "haltAfterReset": true
            },
            "coreConfigs": [
                {
                    // "coreIndex": 0,
                    "programBinary": "${workspaceFolder}/target/thumbv7em-none-eabihf/debug/test-sam",
                    "rttEnabled": true
                }
            ]
        },
```

Things I have verified:
- `which probe-rs-debug` in the VS terminal does come up successfully
- The programBinary binary does exist
- Running the printed command `probe-rs-debugger debug --port 54166 --vscode` works successfully

I'm using workspaces, which I don't think seems to matter.

noppej, 2023-01-25:
@tgross35 How are you able to run while the "type" parameter is commented out? When I do that, VSCode complains that I'm trying to debug "JSON with comments"?

tgross35, 2023-01-25:
I believe vscode uses json5 right? Which accepts trailing commas and comments. At least, I've never had an issue with using json5 syntax

noppej, 2023-01-25:
It is not the comments that is the problem, but rather that you commented out the "type"

tgross35, 2023-01-25:
Whoops sorry, that is just a typo. I fixed the issue

noppej, 2023-01-25:
Can you try this.
1. In the terminal, launch debugger with `probe-rs-debugger debug --port 50001`. It should start it as a server.
2. Then, in your launch.json, add the following property `"server": "127.0.0.1:50001"`

This way, you take care of launching the process, and VSCode just connects to it. It is not a fix, but I want to eliminate variables.

tgross35, 2023-01-25:
Hey, that worked! After setting the `chip` (thought it would maybe pull from Embed.toml). On to an arm error about swj_pins, but I can figure that out.

Anything else I should try to narrow it down further?

noppej, 2023-01-25:
Progress is good. This VSCode debugger is completely different from cargo-embed. The only config happens in launch.json.
(Please update the launch.json above with your current running config)

Back to the original error ... what OS are you running?

tgross35, 2023-01-25:
```json5
       {
            "name": "probe-rs Debug SFP4364",
            "type": "probe-rs-debug",
            // "preLaunchTask": "rust: cargo build sfp-4364",
            "request": "launch",
            "cwd": "${workspaceFolder}",
            // "connectUnderReset": true,
            "server": "127.0.0.1:50001",
            "chip": "ATSAMD51G19A",
            "flashingConfig": {
                "flashingEnabled": true,
                "resetAfterFlashing": true,
                "haltAfterReset": true
            },
            "coreConfigs": [
                {
                    // "coreIndex": 0,
                    "programBinary": "${workspaceFolder}/target/thumbv7em-none-eabihf/debug/sfp-4364",
                    "rttEnabled": true
                }
            ]
        }
```

That is the updated config, I'm on an intel mac

noppej, 2023-01-25:
Everything about that error suggests that it cannot find that file. From the VSCode terminal, can you please do the following: 
`where probe-rs-debugger`
and 
`probe-rs-debugger --version`

tgross35, 2023-01-25:
It definitely should exist an is in the path, so this is quite weird

```
$ where probe-rs-debugger
/Users/tmgross/.cargo/bin/probe-rs-debugger
$ echo $PATH | tr ':' '\n' | grep cargo
/Users/tmgross/.cargo/bin
$ probe-rs-debugger --version
probe-rs-debugger 0.14.2
```

noppej, 2023-01-25:
So weird. Thanks for your patience. If we know the file is there, then it something weird in the way VSCode tries to spawn the process, or something wrong with the executable. 

Can you confirm that your OS is up to date, as well as VSCode?

Were all of these clean installs, or copied from another system?

Can you also do a `cargo install --force probe-rs-debugger`

tgross35, 2023-01-25:
Vscode 1.74.3 and MacOS 12.6.2, fairly up to date. Everything is clean, I haven't had problems with anything else.

I tried the reinstall, same problem :( but I have it working with a separate attach, so that's satisfactory for me for now

noppej, 2023-01-27:
One final test, just to see if the problem lies in "how" VSCode spawns that process, vs. the executable itself (probe-rs-debugger). 

Can you please add the following to your `launch.json`:

`"runtimeExecutable": "/Users/tmgross/.cargo/bin/probe-rs-debugger", // location from your previous feebdack`

noppej, 2023-02-07:
@tgross35 I managed to accidentally reproduce this error, but the condition makes me think that it is not specific enough to be confident that it applies in all cases.

In my case, I had an invalid value for the `cwd` property in `launch.json`. I've just PR'ed a fix to trap that and report the problem, but it might help if you can manually test this on your side, by just removing the `cwd` option from your config.
[probe-rs-debugger-0.4.3.vsix.zip](https://github.com/probe-rs/vscode/files/10676973/probe-rs-debugger-0.4.3.vsix.zip)

closed 2023-02-07 by Yatekii

tgross35, 2023-02-07:
Oh whoops sorry, I missed your last message to try! I'll give it a shot next time I have the debugger with me.

Thanks for all the help!