probe-rs/vscode #42: Release on vscode marketplace

MabezDev, 2023-02-03:
Just wondering if there are any missing features/blockers to be able to push to the market place instead of manually installing from releases :)

Yatekii, 2023-02-03:
I think we could set that up now :) cc @noppej

noppej, 2023-02-03:
I've got a draft for it (haven't uploaded it) in progress. I have to update the graphics/gifs, and need to make one change to make the terminate handling compliant.

Unless I run into serious issues, it should be in the next couple of weeks.

MabezDev, 2023-02-03:
Does it make sense to bundle the latest `probe-rs-debugger` with the extension? I know probe-rs is mostly used for rust projects, but it's also a good replacement for openocd in general, I could see value in having a ready to go setup ootb without needed rust tooling to install stuff.

noppej, 2023-02-03:
I have read some articles about this, but it is a bit tricky for a couple of reasons:
- You have to distribute binaries for all OS'es, which bloat the extension package.
- If the user wants to install an update to probe-rs-debugger with cargo, then you run the risk of having your copy, plus the cargo installed copy. 

Have you seen examples where rust binaries are shipped successfully this way? I'd love to find a way to do it without unlocking a bunch of future headaches.

MabezDev, 2023-02-03:
I don't have any references, but rust-analyzer used to work this way. I don't know if they downloaded the binary for the arch on startup of the extension or shipped with it though.

noppej, 2023-02-03:
Do you know anybody in that community that I might reach out to?

Yatekii, 2023-02-03:
We can just look at the rust-analyzer code :)

noppej, 2023-02-16:
@MabezDev and @Yatekii 

I had a look at what rust-analyzer does. They have a pretty neat setup, that results in releasing 9 platform specific ".vsix" files to the marketplace.

Their release workflows are pretty complex, and more than I'm willing to take on for the scope of the first release to the marketplace. In addition, they have to manage the 'client' and 'server' processes under VSCode to make sure both are running, and talking to each other.

My recommendation is that I finish #45, and then we can consider PR's for bundling the platform specific binaries.

closed 2023-02-21 by noppej