probe-rs/vscode #47: Improve Readme for  published

Tiwalun, 2023-02-21:
Great to see the extensions in the marketplace!

It seems that it now includes the README.md from this repo as the description for the extension.
The readme has a lot of content specific to the development of the extension itself, which seems a bit out of place.

 I don't know how the packaging works, but it would be nice to have more user focused review in the market place, and a developer focused in the repo.

noppej, 2023-02-21:
I think that can be done ... I would probably take this approach:
- marketplace README: just a teaser / brief showcase. 
- GitHub README: same, plus info for contributors 
- probe.rs: full docs.  

Comments?

Yatekii, 2023-02-21:
Sounds like a good plan to follow :)

closed 2023-02-21 by noppej