probe-rs/vscode #56: Save RTT logs

uliano, 2023-03-27:
When I ctrl+C a cargo embed session all the RTT logs are gone.
It would be very nice being able to configure in the launch.json the possibility to save them in either defmt or decoded text,

If a file per each RTT channel is saved on the go it will allow for realtime graphics and other niceties...

uliano, 2023-03-27:
I would say both.

If vscode can intercept the terminal output it could be implemented there whit an easy configuration in launch.json.

Otherwise a new command line option for cargo embed will probably be required.

Yatekii, 2023-03-27:
vscode has no relationship with vscode. So it would be wrong to have vscode do anything with cargo-embed output.

cargo-embed has an option to save log files on exit already. It's just not written live as things happen.

uliano, 2023-03-27:
my cargo embed doesn't seem to have it 

`> cargo embed --help   
Common options used when building artifacts with cargo

Usage: embed [OPTIONS] [config]

Arguments:
  [config]  

Options:
      --chip <chip>
      --probe <PROBE_SELECTOR>  Use this flag to select a specific probe in the list.
                                Use '--probe VID:PID' or '--probe VID:PID:Serial' if you have more than one probe with the same VID:PID.
      --list-chips
      --disable-progressbars    
      --work-dir <WORK_DIR>     Work directory for the command
  -h, --help                    Print help
  -V, --version                 Print version`

closed 2023-04-16 by noppej