probe-rs/vscode #58: defmt logs don't show anywhere

uzytkownik, 2023-04-04:
They are displayed with `probe-run` but when I try to run they are nowhere to be found (not in Terminal, Debug Console or Output).

This may be a documentation error as the page specified to use `rttEnabled` in `configuration` element which is flagged by VSCode as incorrect. I tried following:

```json
    "configurations": [
        {
            "type": "probe-rs-debug",
            "request": "launch",
            "name": "probe-rs Test",
            "cwd": "${workspaceFolder}",
            "connectUnderReset": false,
            "chip": "RP2040",
            "flashingConfig": {
                "flashingEnabled": true,
                "resetAfterFlashing": true,
                "haltAfterReset": true
            },
            "env": {
                "DEFMT_LOG": "info"
            },
            "coreConfigs": [
                {
                    "coreIndex": 0,
                    "programBinary": "./target/thumbv6m-none-eabi/debug/${workspaceFolderBasename}",
                    "rttEnabled": true,
                    "rttChannelFormats": [
                        {
                            "channelNumber": 0,
                            "dataFormat": "Defmt"
                        },
                    ]
                }
            ]
        },
```

noppej, 2023-04-04:
You are correct, the location of the `rttEnabled` entry is incorrect in the [docs](https://probe.rs/docs/tools/vscode/).  

One thing that I struggled with at first, is that the way defmt works, is that the log level is coded into the binary at build time, and that the filter at runtime ( e.g. `launch.json`) cannot include logs which are not already in the binary. It is less of a problem with tools like `probe-run`, because the build and run happens from the same command.

Can you confirm that the DEFMT_LOG environment variable is set at the time that you build your binary?

noppej, 2023-04-04:
See https://github.com/probe-rs/webpage/pull/77 for doc fix.

closed 2023-04-16 by noppej