probe-rs/vscode #59: Timeout while attaching to target under reset

tsadowski, 2023-04-06:
Hello,

I had a working setup for debugging a micro:bit in VSCode. But the I decided to update to 0.18.0 and now I just get:

```
 INFO: Triggered DAP Event: Event {
    body: Some(
        Object {
            "message": String("Timeout while attaching to target under reset. This can happen if the target is not responding to the reset sequence. Ensure the chip's reset pin is connected, or try attaching without reset.\n"),
            "severity": String("error"),
        },
    ),
    event: "probe-rs-show-message",
    seq: 4,
    type_: "event",
}

```

I did not alter my launch configuration. Is there anything I need to add?

Cheers,
Torsten

noppej, 2023-04-06:
There were some changes to the handling of "connectUnderReset".  Can you try it with that setting set to false in the launch.json?

tsadowski, 2023-04-06:
Thank you very much, That was it!

closed 2023-04-06 by tsadowski

noppej, 2023-04-06:
I will update the error message to clarify.