probe-rs/vscode #60: Bundle binary in extension

Tiwalun, 2023-06-12:
It's possible to bundle binaries in a vscode extension, and this can be done in a platform specific way. See https://code.visualstudio.com/api/working-with-extensions/publishing-extension#platformspecific-extensions

It would be nice to bundle the debugger binary, so that users don't have to install a separate tool.

noppej, 2024-01-22:
@Tiwalun Quite a few people I know of, use this extension with `probe-rs` from the `master` branch (usually a few fixes and features ahead of the published). 

By bundling the binary, would we add complications for those use cases?

Tiwalun, 2024-01-22:
Not really, we can just have a configurable setting where you can choose the bundled binary or a different one.

Or you use the existing mode where you connect over tcp.

noppej, 2024-01-22:
Fair. We existing config settings in `launch.json` to overwrite the binary. I think we can re-use that with some adjustment.

noppej, 2024-01-23:
I had a look at the examples included above, and cannot see how to get from their 'dependency' on a node package that includes the platform executables, to our own cargo release of the binaries.  For now, I will leave this enhancement for someone else to attempt, with this final note:

The extension currently supports the `runtimeExecutable ` property in the `launch.json` config. It will be nice if we can leverage that to toggle between the "packaged" versus "locally installed" version of `probe-rs`, rather than introduce a new property.

Tiwalun, 2024-05-09:
> I had a look at the examples included above, and cannot see how to get from their 'dependency' on a node package that includes the platform executables, to our own cargo release of the binaries. For now, I will leave this enhancement for someone else to attempt, with this final note:

cargo dist supports npm packages, so it might be quite easy to get this:

https://opensource.axo.dev/cargo-dist/book/installers/npm.html

Tiwalun, 2024-05-09:
But it seems we could also place a binary in the extension directory, and then bundle it. That might be easier than using a complete other package.

noppej, 2024-05-09:
I'm guessing it is safe to assume we have a binary for each of the VSCode supported platforms?

noppej, 2024-05-22:
@Tiwalun Can we close this as fixed by #93?

closed 2024-05-23 by Tiwalun