probe-rs/vscode #65: Replace the VSCode peripheral viewer with mcu-debug

kossnikita, 2023-10-26:
Currently, the built-in peripheral viewer has less functionality compared to [mcu-debug/peripheral-viewer](https://github.com/mcu-debug/peripheral-viewer).

From what I noticed:
- Enumerations are ignored
- Ignored register size (always 32 bits)
- Writing does not work in my case

Does it make sense to use peripheral-viewer instead? I think the idea behind this extension is very similar to the viewer in your extension. It has been used with [cortex-debug](https://github.com/Marus/cortex-debug) for quite some time, so it will be a little familiar to developers switching to Rust from C.

If replacement is not possible, is it at least possible to transfer the functionality of this extension to your extension?

Extensions comparison
[image]
[image]

kossnikita, 2023-10-26:
I wasn't sure that this relative to this repository, because I didn't find device peripheral logic in it

kossnikita, 2023-10-26:
> I wrote a web based viewer

Is this already published somewhere?

Yatekii, 2023-10-26:
It's a branch in the `probe-rs/cargo-embed` repository :)

noppej, 2023-10-26:
There is no reason why such functionality cannot be delivered as part of the VSCode extension. The VSCode API, and DAP Protocol extensibility, would make it relatively smooth to present it in a integrated view as part of the debug extension. 

I wrote the current integration as a "proof of concept", to judge usage and requirement, and because I use it myself. The total lack of feedback to this point has lowered the priority in terms of my own efforts.

That said, if you @kossnikita (or anybody else) were interested in PR'ing, I'd be happy to review and support.