probe-rs/vscode #69: Failed to update variable during debug session

no111u3, 2023-12-10:
After change some register value during debug session receive message:
```
Failed to update variable: MODER, with new value "10101011111111111111011111111111 @ 0x48000000" :
Please report this as a bug.
```
This happened for every registers and bit fields who allows to change it value.

MCU: STM32G474RET6, Debugger: ST-Link-v3 (Nucleo development board), Probe-rs: v0.21.1

noppej, 2023-12-18:
Yeah, updating register fields is not supported/implemented yet. The message is misleading. In theory, it should be easy to add this support ...
- Need to carefully check errors, because the SVD's are not always accurate about update rules.

#65 is probably a better solution though.

noppej, 2024-01-22:
Closing in favour of #65 ... unless someone comes up with a better solution.

closed 2024-01-22 by noppej