probe-rs/vscode #76: [Feature request] Colorful defmt log

HaoboGu, 2024-01-17:
Hello, I'm using probe-rs-vscode plugin with defmt rtt logger. When I use `probe-rs run` in cli, it prints logs with color:
[image]

But when I do debugging using probe-rs-vscode plugin, the log is printed at the `defmt` tab, but without color: 
[image]

What's I expect:
log with color in defmt tab

noppej, 2024-01-18:
The extension runs `probe-rs` as a background process, and `defmt-decoder` (or maybe one of its dependencies) disables terminal sequences for color codding.  I am sure this is a solve-able problem, but haven't figured out the solution yet.

In the mean time, you can manually run `probe-rs` [as a server](https://probe.rs/docs/tools/debugger/#connecting-to-a-standalone-probe-rs-dap-server-server), and configure the extension to connect to it. When you do, you will see the color coding you expect.

noppej, 2024-01-18:
@Tiwalun and/or @Yatekii  ... My experience with `Node` is very limited, but I suspect it is [this spawn option](https://github.com/probe-rs/vscode/blob/642d3de1406e8dc708712255088f2bb7b14df644/src/extension.ts#L509) that is causing the behaviour with `defmt`. 

Does that make sense to you? Any suggestions on how to mitigate?

PR #79 (2024-01-22): Force color in RTT output when running in background
Fixes #76
closed 2024-01-22 by Yatekii