probe-rs/vscode #85: Failing to connect to server

petermarks, 2024-02-22:
When starting a debug session, I get "A timeout occurred." in the Debug Console almost immediately. In the developer console I see:

```
[Extension Host] probe-rs-debug: Starting as a DAP Protocol server
probe-rs-debug: No logging data will be written because the RUST_LOG environment variable is not set.
probe-rs-debug: Listening for requests on port 46473

[Extension Host] probe-rs-debug: ..Starting session from   :127.0.0.1:44944

[Extension Host] probe-rs-debug: ....Closing session from  :127.0.0.1:44944
probe-rs-debug: CONSOLE: DAP Protocol server exiting
```

It looks like it starts the server on one port, then tries to connect on a different port. Looking at the code, I don't immediately see how that can happen.

probe-rs-debugger: v0.22.4
vs-code: 1.86.2
OS: Linux x64 6.1.67

noppej, 2024-02-22:
Hi Peter,

I don't think that is what happening. What you are seeing is correct, and can be interpreted as :
- Host says it is listening on 46473
- Host says it opens a dedicated session from a client using port 44944
- Host says it closes a session from the same client

There is obviously something else happening, but I'd need to see more of your config (launch.json) to try to identify why you are getting such an unhelpful message.

petermarks, 2024-02-22:
Thanks for the explanation @noppej.

Here's my launch.json:

```json
{
  "version": "0.2.0",
  "configurations": [
    {
      "type": "probe-rs-debug",
      "request": "launch",
      "name": "probe-rs Test",
      "cwd": "${workspaceFolder}",
      "connectUnderReset": true,
      "chip": "RP2040",
      "flashingConfig": {
        "flashingEnabled": true,
        "haltAfterReset": false
      },
      "coreConfigs": [
        {
          "coreIndex": 0,
          "programBinary": "./target/thumbv6m-none-eabi/debug/${workspaceFolderBasename}"
        }
      ]
    }
  ]
}
```

Let me know if you need any further info.

I should add that I'm using NixOS, which often introduces joy and wonder, but I'm not sure that is relevant here as the server seems to be launching fine.

petermarks, 2024-02-22:
@noppej pointed out on chat that connectUnderReset isn't supported on the RP2040, and removing that solved the issue for me. Agreed, a better error message would be helpful.

closed 2024-02-25 by Yatekii