probe-rs/vscode #100: Can't step into standard library when debugging

staticintlucas, 2024-07-01:
If I try to step into any standard library function the source file is not found, which can be pretty annoying:

[image]

Rust standard libraries use the source path translation: `/rustc/<commit-hash>` &rArr; `<sysroot>/lib/rustlib/src/rust`

`rust-analyzer` takes care of this translation automatically here: https://github.com/rust-lang/rust-analyzer/blob/1b283db47f8de1412c851c92bb4ce4ef039ff8ff/editors/code/src/debug.ts#L144

Would it be possible to support the same here too?

noppej, 2024-07-02:
It is definitely possible, and would be a nice addition. The implementation would be in `dap-server` component of the https://github.com/probe-rs/probe-rs repo. I'd be more than happy to help you get started on that code base if you want to submit a PR.

staticintlucas, 2024-07-02:
> The implementation would be in dap-server component of the https://github.com/probe-rs/probe-rs repo.

Ah sorry, didn't realise it was there

> I'd be more than happy to help you get started on that code base if you want to submit a PR.

Yeah I'd be happy to work on a PR if you gave me a few pointer on where to look

noppej, 2024-07-03:
As a starting point, have a look at the DAP specification for [Source](https://microsoft.github.io/debug-adapter-protocol/specification#Types_Source). We construct this in the `dap-server` as part of creating a [StackFrame](https://github.com/probe-rs/probe-rs/blob/fb1ed62584fdfb62eb9fd0db8eddbd171d28ffa3/probe-rs-tools/src/bin/probe-rs/cmd/dap_server/debug_adapter/dap/request_helpers.rs#L273) from the stack trace information.

My first instinct suggests that you will need to specify a `sourceReference` in the `StackFrame` definition, so that it forces the client to issue a `Source` request, where the 'translation' you refer to above, will be implemented.

Once you've had a look at that, the next step I'd suggest is to create a PR in the `probe-rs` repo, and then tag me from there to continue the conversation in more detail.

closed 2024-07-31 by bugadani