probe-rs/vscode #105: Probe_rs OS Error

Admsher, 2024-08-13:
I have been using arm-none-eabi-gdb to flash my bin file using probe-rs but when i try to connect on port:3333 i get this error message:

GDB:
Remote replied unexpectedly to 'vMustReplyEmpty': timeout

Probe-rs:
ERROR probe_rs::cmd::dap_server::debug_adapter::protocol:24 

The board kit is IMXRT1064 and I am using a DAP server

Yatekii, 2024-08-15:
GDB cannot use the DAP server :) You need the GDB server for it.

Admsher, 2024-08-15:
Oh the documentation did not mention that is there any way I can use probe rs for USB debugging??

noppej, 2024-08-15:
If you specifically want GDB, and not use the probe-rs implementation of DAP for debugging, then perhaps consider the excellent extension at https://github.com/Marus/cortex-debug.

closed 2024-08-15 by noppej