probe-rs/vscode #108: VSCode Extension: Debugger: Disassembly View highlights wrong line

fg-cfh, 2024-12-19:
### Describe the bug

See title and screenshot:

[image]

The PC register shows the correct instruction address. The Disassembly View highlights the wrong address.

Identifier probe-rs.probe-rs-debugger
Version 0.24.2
Last Updated 2024-11-03, 03:19:42

### Stacktrace

```shell

```

### Operating System

Linux (Ubuntu)

fg-cfh, 2024-12-19:
Seems like one possible cause could be the erroneous decoding of instructions, see the interspersed ".byte" instructions, these should actually be 32-bit Thumb instructions. Following the corresponding objdump output of some of the above lines:

                dcdc: DcdcConfig {
    28e2:       f88d 0005       strb.w  r0, [sp, #5]
    28e6:       f89d 2008       ldrb.w  r2, [sp, #8]
    28ea:       f88d 2007       strb.w  r2, [sp, #7]
    28ee:       f88d 0006       strb.w  r0, [sp, #6]

Yatekii, 2024-12-19:
Hmm, do we feed the disasm to the disasm view or does it do that from the ELF?

noppej, 2024-12-19:
> Hmm, do we feed the disasm to the disasm view or does it do that from the ELF?

@Yatekii It requests this through the DAP api. Currently we read the memory addresses it requests from the target, and disassemble those. An alternate approach would be for us to read the ELF instead, as long as we can account for potential memory mapping / relocation after the elf has been loaded.

fg-cfh, 2024-12-20:
> An alternate approach would be for us to read the ELF instead

FWIW: I tested this out of interest. I found that from an ARM ELF the encoding can be identified via [symbol address](https://github.com/pacak/cargo-show-asm/pull/350/files) (ST_VALUE) bit zero (0: Thumb-.v2, 1: Arm) or by [parsing the ".ARM.attributes" section](https://github.com/fg-cfh/cargo-show-asm/commit/5c4d8d7b0c103bb677c16a5042b30e2dbb3f51a1). The former only works when disassembling each symbol separately, the latter only works with uniformly encoded files (not using BX/BLX/...). Once the instruction set is known, Capstone correctly disassembles the code. Are you aware of other/simpler methods?

fg-cfh, 2024-12-20:
> it has no accommodation for the fact that Thumb v2 has variable length instructions (16- and 32-bit)

Hrm, seems like [disassembled instructions](https://microsoft.github.io/debug-adapter-protocol/specification#Types_DisassembledInstruction) could be variable length. Wouldn't that indicate that the actual problem lies in the DAP protocol implementation and should be fixed there? Other clients would otherwise see the same error.

noppej, 2024-12-21:
It's been a while since I worked on this code, but I think you are on the right track. PR's absolutely welcome :)