probe-rs/vscode #109: no-semihosting error

eighteen-k-gold-malow, 2024-12-25:
### Describe the bug

I am using st-link. Why does the probe-rs plugin of vscode use F5 debugging and prompt a `no-semihosting` error when it encounters `hprintln!`, but I can print text normally when I use `probe-rs run --chip stm32f103c8` to execute `hprintln!`

### my code
```rust
hprintln!("I2C write failed: {:?}", e)
```
### config
```json
{
    "version": "0.2.0",
    "configurations": [

        {
            "type": "probe-rs-debug",
            "request": "launch",
            "name": "probe-rs Test",
            "cwd": "${workspaceFolder}",
            "connectUnderReset": false,
            "chip": "stm32f103c8",
            "probe": "0483:3748",
            "flashingConfig": {
                "flashingEnabled": true,
                "haltAfterReset": false
            },
            "coreConfigs": [
                {
                    "coreIndex": 0,
                    "programBinary": "./target/thumbv7m-none-eabi/debug/${workspaceFolderBasename}"
                }
            ]
        }
    ]
}
```

### Operating System

win11-x64

### This is the normal code
```
PS C:\Users\xxx\Desktop\app> probe-rs run --chip stm32f103c8 ./target/thumbv7m-none-eabi/debug/app
      Erasing ✔ 100% [####################]  33.00 KiB @  33.43 KiB/s (took 1s)
  Programming ✔ 100% [####################]  33.00 KiB @  14.11 KiB/s (took 2s)                                                          
     Finished in 2.34s
I2C write failed: Acknowledge
```

###This is the exception code in the debugger
[image]
[image]

Yatekii, 2024-12-25:
We dont really recommend using semihosting (for printing) and probe-rs-tools only recently got support for it. VSCode never got retrofitted and imo we should not support it to actively discourage it :)

eighteen-k-gold-malow, 2024-12-25:
> We dont really recommend using semihosting (for printing) and probe-rs-tools only recently got support for it. VSCode never got retrofitted and imo we should not support it to actively discourage it :)

I am not very clear about this. I have just started learning STM32. If the VSCode plugin does not support it, how can I print debugging information

Yatekii, 2024-12-25:
With the `rtt-target` crate and RTT :)

closed 2024-12-25 by eighteen-k-gold-malow