probe-rs/vscode #110: It is recommended to remove this plugin

eighteen-k-gold-malow, 2024-12-29:
Even if all optimizations are turned off in debug mode, the generated code is inconsistent with the debug code. Sometimes breakpoints are set on each line, but none of them take effect. Sometimes breakpoints take effect, but the code inside the library is automatically entered, and the variables may not be displayed. I think there is no need to develop it. Anyway, there is already a good vscode plug-in `Cortex-Debug` on the market.

Yatekii, 2024-12-29:
You will have exactly the same issues with breakpoints and debug symbols with cortex debug as they use the same chip resources and debug symbols. Symbol resolution is arguably worse even.

If you are having issues with the plugin it would be better to open specific issues so we can address the issues. Cortex Debug works with other debug toolkits that require elaborate setups and offer a different experience :)

closed 2024-12-29 by Yatekii

eighteen-k-gold-malow, 2024-12-29:
> You will have exactly the same issues with breakpoints and debug symbols with cortex debug as they use the same chip resources and debug symbols. Symbol resolution is arguably worse even.
>
> If you [...]

Isn't the debugging symbol generated by Rust? When I debug main.rs, this plugin often jumps to other Rust libraries to execute internally