probe-rs/vscode #111: defmt: no location information

erichstuder, 2025-01-04:
Hello

I have an odd behavior I wasn't able to solve so far.

When I use the [Debugger for probe-rs](https://marketplace.visualstudio.com/items?itemName=probe-rs.probe-rs-debugger) in VS Code then after flashing a terminal window opens showing the defmt messages but without location information (file and function names and line numbers, ...)

```
0.010634 INFO  USB: config_descriptor used: 70
0.010900 INFO  USB: bos_descriptor used: 12
0.011067 INFO  USB: msos_descriptor used: 0
0.011208 INFO  USB: control_buf size: 64
```

my launch.json:

```
{
    "version": "0.2.0",
    "configurations": [
        {
            "name": "probe-rs-debug",
            "type": "probe-rs-debug",
            "request": "launch",
            "preLaunchTask": "firmware_build",
            "cwd": "${workspaceFolder}/firmware",
            "connectUnderReset": false,
            "chip": "RP2040",
            "flashingConfig": {
                "flashingEnabled": true,
                "haltAfterReset": true,
            },
            "coreConfigs": [
                {
                    "coreIndex": 0,
                    "programBinary": "build/thumbv6m-none-eabi/debug/firmware",
                    "rttEnabled": true
                }
            ]
        }
}
```

But if I run probe-rs directly from command line, the location information is there.
So I think everything is basically there.
Is this a limitation of the VS Code plugin or did I miss something?

```
erich@23f312b4f076:~/software$ probe-rs run --chip RP2040 firmware/build/thumbv6m-none-eabi/debug/firmware
      Erasing ✔ 100% [####################] 484.00 KiB @  50.93 KiB/s (took 10s)
  Programming ✔ 100% [####################] 484.00 KiB @  14.54 KiB/s (took 33s)                                                                               Finished in 33.30s
0.010890 INFO  USB: bos_descriptor used: 12
└─ embassy_usb::builder::{impl#1}::build @ /usr/local/cargo/registry/src/index.crates.io-6f17d22bba15001f/embassy-usb-0.3.0/src/builder.rs:199 
0.011069 INFO  USB: msos_descriptor used: 0
└─ embassy_usb::builder::{impl#1}::build @ /usr/local/cargo/registry/src/index.crates.io-6f17d22bba15001f/embassy-usb-0.3.0/src/builder.rs:200 
0.011222 INFO  USB: control_buf size: 64
└─ embassy_usb::builder::{impl#1}::build @ /usr/local/cargo/registry/src/index.crates.io-6f17d22bba15001f/embassy-usb-0.3.0/src/builder.rs:201 
```

noppej, 2025-01-04:
There were some changes made in PR #90 that didn't make it into the docs as far as I can tell.  Try one of these options in your `launch.json` ...

```json
"rttEnabled": true,
"rttChannelFormats": [
  {
    "channelNumber": 0,
    "dataFormat": "Defmt",
    "showLocation": true,
    "showTimestamps": true
 }
```

or adding a format string from the many options documented by [defmt](https://defmt.ferrous-systems.com/custom-log-output)

```json
"rttEnabled": true,
"rttChannelFormats": [
  {
    "channelNumber": 0,
    "dataFormat": "Defmt",
    "logFormat": "{L} {s}"
 }

```

erichstuder, 2025-01-04:
Works.
Thank you!

noppej, 2025-01-04:
Glad to hear.  

Re-opening as a documentation bug :-)