probe-rs/vscode #112: Add support for 64bits values in variable viewer

igiona, 2025-01-16:
Currently, 64bits values aren't shown, instead the following error is displayed

```
Unimplemented: Get value of type Modified(Typedef("uint64_t"), Modified(Typedef("__uint64_t"), Base("long long unsigned int"))) of (Some(8) bytes) at location 0x200013B0
```

noppej, 2025-01-16:
This should be a relatively easy functionality addition for PR. I'd be happy to provide guidance if you are interested in taking it on :)

igiona, 2025-01-16:
Yes, I was planning to have a look at it but I first need to fill the gap in the VsCode-extensions world.
I have only a fuzzy memory-map on how things play together in VsCode

Do you have some resources on this regard?

noppej, 2025-01-16:
Actually, in this case, it would require zero changes to the VSCode extension. All the code will be in the probe-rs debugger code (Rust). When you're ready to jump in, just DM me on matrix, and we can do a quick session for me to walk you through the code layout, and how to debug the debugger code for stepping, etc.

noppej, 2025-01-16:
> Do you have some resources on this regard?

The probe-rs VSCode extension is just a very thin layer that manages to connection the probe-rs debugger backend. The backend is implemented in Rust and communicates with the extension using the [DAP Protocol](https://microsoft.github.io/debug-adapter-protocol/).

If you haven't discovered it yet, there is also some [probe-rs specific docs](https://probe.rs/docs/tools/debugger/), although it is written mostly for "how to use" and could do with some "how to get started if you want to add functionality"  perspective.