probe-rs/vscode #114: VSCode probe-rs extension don't highlight (change highlighting/coloring) of Rust source code line at which breakpoint has triggered

jerrywrice, 2025-02-17:
Using VSCode (version 1.97.2) with probe-rs-debug extension (version 0.24.2) doesn't visually indicate the specific rust source file line of code at which a breakpoint is initially triggered, nor when single-stepping.

noppej, 2025-02-22:
@JustinG002 Thank you for providing a test case!

Can you please confirm the versions of probe-rs and the VSCode extension ?

Yatekii, 2025-02-22:
For reproduction it would also be great to have a RAM dump of the run.

noppej, 2025-02-22:
@Yatekii the runtime instruction to file/line/column seems to be successful, so I suspect the problem is not in the target, but in probe-rs resolving the file paths.   I will look into it when I have version information.

JustinG002, 2025-02-22:
@noppej Thanks for the quick response, See version information below
- probe-rs: 0.27.0
- vscode: 1.97.2
- rustup: 1.27.1
- arm toolchain: 14.2.Rel1

Let me know if you are interested in more information.

Yatekii, 2025-02-22:
Yep, but with the RAM dump we can reproduce the issue :)

JustinG002, 2025-02-22:
@Yatekii Could you explain how would I do that?

Yatekii, 2025-02-22:
You can use the `dump` command in the repl :)

JustinG002, 2025-02-22:
Please see the elf file used in the test above
https://github.com/JustinG002/file_sharing/blob/66d64f188717793ed61378087db74089292ed88e/thing

mailiam, 2025-04-05:
How's this going? I just started my journey on embedded rust and I'm stuck on this issue.
My environment is:

```
probe-rs: 0.27.0
probe-rs.probe-rs-debugger: 0.24.2
```

The project was setup with `esp-generate --chip esp32c3 -o unstable-hal -o alloc -o wifi -o ble -o embassy -o probe-rs -o defmt -o embedded-test -o ci -o vscode`

also attaching elf and coredump(`dump 0 65536 issue-114.dump`)

[issue-114.zip](https://github.com/user-attachments/files/19616835/issue-114.zip)

[image]